thegecko/web-bluetooth-dfu #6: Writing DFU Control Point Characteristic causes disconnect with no write response

beaufortfrancois, 2016-03-01:
As you can see from these logs, when enabling notifications and writing `01 04`, I got no write response with https://thegecko.github.io/web-bluetooth-dfu/firmware/nrf51_boot_s110.hex device.

I believe this is a firmware issue.
cc @mjdietzx 

```
< ACL Data TX: Handle 32 flags 0x00 dlen 9                                                                                                                [hci0] 307.414723
      ATT: Write Request (0x12) len 4
        Handle: 0x000f
          Data: 0100
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                    [hci0] 307.477326
        Num handles: 1
        Handle: 32
        Count: 1
> ACL Data RX: Handle 32 flags 0x02 dlen 5                                                                                                                [hci0] 307.544193
      ATT: Write Response (0x13) len 0

< ACL Data TX: Handle 32 flags 0x00 dlen 9                                                                                                                [hci0] 316.279729
      ATT: Write Request (0x12) len 4
        Handle: 0x000e
          Data: 0104
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                    [hci0] 316.320261
        Num handles: 1
        Handle: 32
        Count: 1
> HCI Event: Vendor (0xff) plen 7                                                                                                                         [hci0] 316.793279
        82 12 95 40 10 36 59                             ...@.6Y         
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                            [hci0] 316.794252
        Status: Success (0x00)
        Handle: 32
        Reason: Connection Timeout (0x08)
@ Device Disconnected: F1:6F:DE:EC:F1:D6 (2) reason 1

thegecko, 2016-03-01:
@beaufortfrancois, Is this also apparent with the firmware shared by @mjdietzx?

beaufortfrancois, 2016-03-01:
Nope. It works perfectly fine with his "recent" firmware.

beaufortfrancois, 2016-03-01:
Here's log with @mjdietzx firmware:

```
< ACL Data TX: Handle 32 flags 0x00 dlen 9                                                                                                                [hci0] 879.625841
      ATT: Write Request (0x12) len 4
        Handle: 0x001e
          Data: 0100
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                    [hci0] 879.628758
        Num handles: 1
        Handle: 32
        Count: 1
> ACL Data RX: Handle 32 flags 0x02 dlen 5                                                                                                                [hci0] 879.695512
      ATT: Write Response (0x13) len 0

< ACL Data TX: Handle 32 flags 0x00 dlen 9                                                                                                                [hci0] 887.398931
      ATT: Write Request (0x12) len 4
        Handle: 0x001d
          Data: 0104
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                    [hci0] 887.458731
        Num handles: 1
        Handle: 32
        Count: 1
> ACL Data RX: Handle 32 flags 0x02 dlen 5                                                                                                                [hci0] 887.525854
      ATT: Write Response (0x13) len 0
> HCI Event: Vendor (0xff) plen 7                                                                                                                         [hci0] 887.661720
        82 12 95 40 10 36 59                             ...@.6Y         
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                            [hci0] 887.662709
        Status: Success (0x00)
        Handle: 32
        Reason: Remote User Terminated Connection (0x13)
@ Device Disconnected: F1:6F:DE:EC:F1:D6 (2) reason 3

beaufortfrancois, 2016-03-01:
For info, I've filed a chromium bug as well: https://bugs.chromium.org/p/chromium/issues/detail?id=591069

thegecko, 2016-03-01:
OK, I can raise this with the mbedOS team. I'm also happy to swap the example on here to use @mjdietzx's firmware if it hasn't got this error. What do you guys think?

beaufortfrancois, 2016-03-02:
I would be happy if you can talk to the mbedOS team about this.
Regarding @mjdietzx's firmware, I'll let him chime in ;)

mjdietzx, 2016-03-02:
Yep, I looked into this issue and also believe it is a firmware issue as I tested this with Nordic's BLE tools (Master Control Panel) and saw the same behavior. We can either add a workaround (a bit hacky) in the dfu.js where we resolve after writing that characteristic without checking it was successful (and assume the remote BLE device restarts in DFU mode) or we can swap in the nRF51 I provided.

Shouldn't be a difficult fix on the mbed OS side but I'm sure it would be helpful to them if we reported it.

thegecko, 2016-03-02:
Have raised this issue with the team.

beaufortfrancois, 2016-03-02:
Thanks! Is there a ticket/issue we can follow?

thegecko, 2016-03-02:
Not publicly unfortunately.

thegecko, 2017-05-24:
Closing this issue as it's not a problem with this library.

closed 2017-05-24 by thegecko