thegecko/web-bluetooth-dfu #8: Updating the SoftDevice and or Bootloader via DFU not working

mjdietzx, 2016-03-02:
I believe this is because of the way the hex files are converted to binaries. My guess is a lot of padding is added and this padding is actually sent for the update. Currently looking into this.

mjdietzx, 2016-03-02:
currently in hex2bin() if there are any jumps in the hex file we will get an error.

thegecko, 2016-03-02:
This was a basic implementation I pulled together to get this project off the ground, perhaps a more comprehensive conversion routine is required.

mjdietzx, 2016-03-02:
Yeah, I thought so. I'm going to start working on making hex2bin() more comprehensive. there are some notes here: https://github.com/NordicSemiconductor/nRF-Master-Control-Panel/blob/master/init%20packet%20handling/How%20to%20generate%20the%20INIT%20file%20for%20DFU.pdf

I'll keep hex2bin() general. then based on the ImageType we will have to modify the binary.
Application: Nothing needs to be done (unless it writes to UICR maybe)
SoftDevice: Must remove the Master boot record from binary data as this isn't ever updated (0x0-0x1000)
Bootloader: This writes to UICR so slight modification has to be made to the firmware image.
softDevice + bootloader: (no idea how to do this yet..)

thegecko, 2016-03-02:
Nice! I look forward to it

mjdietzx, 2016-03-10:
#11 Implemented in https://github.com/thegecko/web-bluetooth-dfu/pull/11 and merged

closed 2016-03-10 by mjdietzx