thegecko/web-bluetooth-dfu #9: For some versions of nRF51 we may be restricted with the type of updates we can do

mjdietzx, 2016-03-02:
There are two types of nRF51 devices. One has 128kb flash, the other has 256 kb flash.

When updating a softdevice, bootloader, or softdevice+bootloader we MUST always do a dual bank update (original image is not erased until we have downloaded the entire new image and validated we received it properly). So we need at least twice (but actually a bit more) available on the device of the image we want to transfer. (Note: the application on the device is always erased when updating the softdevice and or bootloader).

There will be cases for both 128kb and 256kb where we can't perform a DFU update. I'm not sure what kind of error code will be returned in this case (as at some point we should get an error in the OPCODE) but this should be tested and maybe we have to do some extra error checking/handling.

Still yet to evaluate this.

thegecko, 2017-05-24:
@mjdietzx Is this issue now defunct due to the release of the new Secure DFU transport prorocol?

thegecko, 2017-06-05:
Closing due to inactivity, please re-open if still relevant.

closed 2017-06-05 by thegecko