thegecko/web-bluetooth-dfu #12: How to implement a seamless complete DFU

mjdietzx, 2016-03-08:
1. Switch from application mode to DFU mode.
2. Disconnect.
3. Transfer image and complete the DFU procedure.

I am having problems getting these steps to run seamlessly. See how both() is implemented in example_web.html: https://github.com/thegecko/web-bluetooth-dfu/blob/master/example_web.html#L103.

The problem I see is at the highlighted line. Here we are reusing device and we are carrying over the old GATT information. This leads to having all the services the application had even though DfuTarg may not have them. I think this may cause problems with attribute handles getting mixed up and I am seeing errors (although if I step this line: https://github.com/thegecko/web-bluetooth-dfu/blob/master/dist/dfu.js#L219 while debugging the entire process can work).. I think this will lead to other problems as well in the future and carrying over the device we discovered when setting mode and just reusing it seems like a hack.

The obvious solution I've tried is to rediscover DfuTarg after setMode() but this leads to a security error being thrown, something like: 'security error: discovering devices requires user input.'

I don't think we can use Service Changed Indications and carry over 'device' from setMode() because the remote BLE device disconnects and resets (and we will lose indications just like we lose CCCD notifications unless we are bonded). Then we are right back where we started.

Do you guys have any ideas about this? Any better solutions or suggestions on how to implement this?

mjdietzx, 2016-03-08:
@beaufortfrancois

beaufortfrancois, 2016-03-08:
ccing @jyasskin @scheib @g-ortuno

thegecko, 2016-03-08:
It's with noting the node example version of this does re-discover the device so doesn't see this problem.

Should the BLE caches be cleared after a disconnect perhaps?

mjdietzx, 2016-03-08:
Re-discovering after receiving a disconnection from the application is something I have tried and think would be a good solution. When I try this though I get 'SecurityError: Must be handling a user gesture to show a permission request.'

beaufortfrancois, 2016-03-08:
For the sake of completeness, here are trace logs from Chrome OS Bluetooth Daemon in debug mode http://pastebin.com/raw/mHpnMwu5 and logs from Chrome Bluetooth vmodule http://pastebin.com/raw/UKQHrUHp

g-ortuno, 2016-03-08:
> This leads to having all the services the application had even though DfuTarg may not have them.

How are you getting the services? Are you caching them yourself or calling getPrimaryService(s)? There is no code in Chrome to cache the services. Whenever you call getPrimaryService(s) we always ask bluez what services exist. If the devices is not bonded and bluez is caching services then it's a bluez bug that we need to get fix.

> I think this may cause problems with attribute handles getting mixed up and I am seeing errors 

What errors are you seeing?

jyasskin, 2016-03-08:
If the device is re-using attribute handles to refer to different services, that's a device bug: 3.F.3.2.2 says "An attribute handle shall not be reused while an ATT Bearer exists between a client and its server."

If the device changes the contents of a service without indicating ServiceChanged to bonded devices, that's also device bug: 3.G.7.1 says "shall".

If bluez is caching services for non-bonded devices across re-connect, that's a bluez bug.

Which is DFU doing?

jyasskin, 2016-03-08:
I only see one "connect" in http://pastebin.com/raw/UKQHrUHp. @g-ortuno, could this be us not handling remote disconnection?

"SecurityError: Must be handling a user gesture to show a permission request." happens when you call `requestDevice()`, which isn't a way to force Chrome to re-discover services on a device you already have.

g-ortuno, 2016-03-08:
I don't see a disconnect either, so from those logs it seems bluez doesn't even know the device disconnected.

jyasskin, 2016-03-08:
I see it in http://pastebin.com/raw/mHpnMwu5, the bluez log. It's just not making it to Chrome.

beaufortfrancois, 2016-03-08:
I'm not able to see remote gatt disconnection in Chromium source code.
Where should I be looking?

g-ortuno, 2016-03-08:
https://code.google.com/p/chromium/codesearch#chromium/src/device/bluetooth/bluetooth_device_bluez.cc&sq=package:chromium&type=cs&l=760

g-ortuno, 2016-03-08:
ah never mind that's only for user initiated disconnection

g-ortuno, 2016-03-08:
@jyasskin: I don't think it's something on our side. We never keep track of the connection state ourselves, we always query bluez for the device's connection state.

beaufortfrancois, 2016-03-09:
@mjdietzx Can you confirm that you're not actually running https://github.com/thegecko/web-bluetooth-dfu/blob/a4d1e684fe0feb3bc2c6890f2d4393ef68e8d16d/dist/dfu.js#L118 as the `gattserverdisconnected` event is not **yet** triggered.
This means you won't get the `resolve('device')` call.

mjdietzx, 2016-03-09:
> This means you won't get the resolve('device') call.

@beaufortfrancois I add resolve(device); after writing the control characteristic succeeds - just locally when I test this. So I'm not relying on the gattserverdisconnect  event.

Looking into the rest of your guys' comments now.

beaufortfrancois, 2016-03-09:
You may want to add a small delay (around 5sec) just to be sure.

```javascript
    return new Promise(function(resolve) {
      setTimeout(function() {
        resolve(device);
      }, 5e3);
    })

mjdietzx, 2016-03-09:
> How are you getting the services? Are you caching them yourself or calling getPrimaryService(s)? There is no code in Chrome to cache the services. Whenever you call getPrimaryService(s) we always [...]

@g-ortuno: How I am getting services:
```javascript
device.gatt.connect()
.then(function(gattServer) {
    return gattServer.getPrimaryService(SERVICE_UUID);
})
```
I am not caching them. **However** as I said above we are reusing 'device.'
```javascript
dfu.findDevice({ services: [0x180D] })
.then(device => dfu.writeMode(device))
.then(device => transfer(device))
.catch(log);
```
Looking at the code above we first discover a heart rate monitor device. This heart rate monitor device has a DFU service. We are able to use this DFU service to reset the device into bootloader mode. We write a control characteristic in the DFU service and the device resets and restarts a completely different application. This new application has different GAP settings and a different GATT profile hierarchy. The similarities between the two applications are:

1.) DFU/bootloader mode Bluetooth device address = Application mode Bluetooth device address + 1.
2.) They both contain the DFU service in their GATT profile hierarchy.

And this leads to where I think problems start to happen. Our 'device' instance has been configured and initialized when we discover the heart rate application and then our DFU example saves this state and (somehow?) is able to connect to this device and discover the DFU service and its characteristics when it restarts into bootloader mode.

Now that you understand exactly what we are doing,

>  What errors are you seeing?

- When running normally 'NotSupportedError: GATT operation failed for unknown reason.' occurs when we try the first GATT procedure: versionChar.readValue();
- When debugging the entire process can work if we set a breakpoint before discovering the primary service.

As I said reusing 'device' seems like a hack and I think it is causing these problems. I don't think the problem is caching anymore. But what other options do we have for performing a complete DFU? In the original issue I explain why we can't rediscover the dfu/bootloader mode device or the use service changed indication.

beaufortfrancois, 2016-03-09:
My 2 cents there: Reusing device should not cause issues. We need to make sure it works even after a disconnection.

mjdietzx, 2016-03-09:
@jyasskin:

> If the device changes the contents of a service without indicating ServiceChanged to bonded devices, that's also device bug: 3.G.7.1 says "shall".

Does Web Bluetooth support bonding? I thought all it did was pair with devices. If bonding is supported we can use service changed indications, otherwise they won't work for this use case.

> If bluez is caching services for non-bonded devices across re-connect, that's a bluez bug.

I don't think this is the problem.

> "SecurityError: Must be handling a user gesture to show a permission request." happens when you call requestDevice(), which isn't a way to force Chrome to re-discover services on a device you already have.

If you see my above comment, do you still think we 'have' the device ? I'm wondering about the gatt propery of device

> gatt provides a way to interact with this device’s GATT server.

mjdietzx, 2016-03-09:
@beaufortfrancois: I am using the timeout in all testing as you suggested above. You can see the exact code I am using at this branch: https://github.com/mjdietzx/web-bluetooth-dfu/tree/refactor. It is the same as thegecko/web-bluetooth-master except it doesn't rely on the disconnect event to resolve writeMode() and it has a 5 second timeout after switching from application to dfu mode.

g-ortuno, 2016-03-09:
> When debugging the entire process can work if we set a breakpoint before discovering the primary service.

This leads me to believe that the bluez is operating with a out of date cache so when you try to read/write it fails.Adding the breakpoint gives bluez enough time to update its cache. Could you try adding a timeout after connection?

mjdietzx, 2016-03-09:
> This leads me to believe that bluez is operating with an out of date cache so when you try to read/write it fails. Adding the breakpoint gives bluez enough time to update its cache. Could you try adding a timeout after connection?

This fixed it! Entire DFU process now works after adding a delay after calling device.gatt.connect() and before discovering the DFU service. https://github.com/mjdietzx/web-bluetooth-dfu/commit/44539fba63ab8d5844b13480a02af5606244838b

thegecko, 2016-03-09:
@mjdietzx Will this be 'automagically' fixed when the ```disconnect``` event appears or should we be rolling this delay into the script?

g-ortuno, 2016-03-09:
I think this is a problem with bluez's cache. Adding the disconnect event won't help.

mjdietzx, 2016-03-09:
@thegecko: The disconnect g-ortuno suggested that fixes the problem with bluez's cache will need to stay.

The timeout that we have in writeMode() we should be able to remove when the disconnect event is implemented.

This delay after calling device.gatt.connect() is actually solving almost all the problems I've been seeing! So this is a huge help! DFU is working very well now.

mjdietzx, 2016-03-09:
Actually the timeout in writeMode isn't needed at all. Works fine just resolving after writing the control characteristic.

beaufortfrancois, 2016-03-09:
@g-ortuno Should we ask BlueZ folks for a fix? Is a fix needed?

g-ortuno, 2016-03-09:
@beaufortfrancois I would wait for the new Bluez version. Maybe it's fixed there. Otherwise, yes we need to ask for a fix.

beaufortfrancois, 2016-03-09:
Logs I provided are from yesterday's master.

g-ortuno, 2016-03-09:
Then yes. We should ask for a fix.

beaufortfrancois, 2016-03-09:
See http://permalink.gmane.org/gmane.linux.bluez.kernel/66940

jyasskin, 2016-03-09:
Nice work @g-ortuno!

To answer questions so we're all on the same page: 
* Web Bluetooth doesn't provide any control over bonding, but platforms may automatically bond in Just Works mode when you connect a device. Devices that can bond at all, need to work in both modes.
* After disconnecting, whether the page does it by calling `.disconnect()` or the remote device does it by physically disconnecting, as @beaufortfrancois said, you still "have" the device, meaning both the `BluetoothDevice` object and permission to communicate with its GATT server. You have to re-connect to actually communicate, but you haven't dropped any permissions. You _shouldn't_ have to manually wait for bluez to re-discover services, but clearly you do until we get bluez fixed.

beaufortfrancois, 2016-03-10:
You can follow BlueZ discussion at http://comments.gmane.org/gmane.linux.bluez.kernel/66940

beaufortfrancois, 2016-03-10:
For info, I've tried this patch on Android and it fails. See adb logs at http://pastebin.com/raw/kPyhzGRB
cc @g-ortuno

Vudentz, 2016-03-10:
So it seems void BluetoothDispatcherHost::OnGetPrimaryService is buggy:

  std::vector<BluetoothGattService*> services =
      GetPrimaryServicesByUUID(query_result.device, service_uuid);

  // 1. & 2.
  if (!services.empty()) {

This appears to be assuming if the services is not empty they should be valid which is not always true because gatt_services_ is not cleanup on disconnection in case of BlueZ, nor it should be because the object can still provide useful information and some functionality still work like stopping notifications even when offline. So we will have to do changes for the BlueZ backend in order for this to work properly, most likely I will propose introducing a new property to track the discovery progress.

beaufortfrancois, 2016-03-10:
Thanks Luiz.
How does http://comments.gmane.org/gmane.linux.bluez.kernel/66963 sound to Web Bluetooth folks?

Vudentz, 2016-03-10:
@g-ortuno Im afraid what you are suggesting has the same result, it just wait until the service discover is completed but in a different layer, but BlueZ don't actually have any API to actively fetch services like other stacks may have thus why BlueZ backend just caches the objects and Im surprised nobody realized that up until now. Now the changes Im proposed to BlueZ may not affect other systems because it seems it is only BlueZ that try to make a cache of services which is done exactly to mitigate the cost of discovering on each and every reconnection so it beats me why this would be a problem in case of BlueZ where for every other stack else it isn't.

g-ortuno, 2016-03-10:
@Vudentz I see. I was under the impression that Bluez worked like CoreBluetooth and that we could access a discovered Service before full discovery.

IIUC, your proposed change would make it so after connecting the `Discovering` property would be `true` and after all services have been discovered the property would change to `false`. If the device emits a Service Changed then the `Discovering` property would change to `true` again and after discovery is complete it would change to `false`. If this is the case then your proposal works perfectly for us.

thegecko, 2016-03-10:
@mjdietzx I've merged your PR containing the timeouts and updated it to work end-to-end with the node example. Can I ask which version of ChromeOS you have this working on so I can test it?

mjdietzx, 2016-03-11:
@thegecko I am using version 50.0.2657.0 dev-channel swanky. Seeing now that there is a new update so I'll test with latest.

thegecko, 2016-03-11:
Success!
Current master branch works using the node/bleat combo and with ChromeOS dev-channel.
Now to get it working in Android, Mac, Windows and Linux :)

Using the mbed image, I couldn't get the full update to work on ChromeOS due to #6, only in parts (it works on node, however).

I'll add Nordic's nrf51 image to the repo too while we get this fixed.

Vudentz, 2016-03-11:
@g-ortuno I wonder how it would be possible to access discovered services before full discovery, at least with the current Web Bluetooth API that doesn't seems to be possible since you get everything at once, at least on chrome that the impression I got from BluetoothHostMsg_GetPrimaryService.

Regarding the discovering, yes it should toggle whenever a GATT discover procedure is in progress, but I don't think it would work for Service Changed when that is indicated outside of the getPrimaryService since there is no way to notify about any service removed/added, so perhaps some sort of service notification mechanism should be added.

g-ortuno, 2016-03-11:
We do have a way of notifying of service changes https://webbluetoothcg.github.io/web-bluetooth/#service-change-events

Vudentz, 2016-03-12:
@g-ortuno Intestings, interesting so applications should be able to detect service changes with service_added, service_removed, etc: https://webbluetoothcg.github.io/web-bluetooth/#eventdef-bluetoothremotegattservice-serviceadded

But then I wonder why you were suggesting to delay the response to getPrimaryService if changes can be handled later? Specially in case the application like this that knows there will be changes to the database it shall subscribe to these events.

jyasskin, 2016-03-13:
We should probably take the BlueZ/Chromium discussion to another thread. Maybe https://groups.google.com/a/chromium.org/d/topic/web-bluetooth/PQpZAFNlvT4/discussion?

thegecko, 2016-03-13:
In that case, @mjdietzx is our work on this issue done?

closed 2016-03-13 by mjdietzx