thegecko/web-bluetooth-dfu #18: DFU transfer often stalls and breaks while sending the FW image

mjdietzx, 2016-03-17:
Now everything is working pretty stable when switching to DFU mode and initializing the transfer. But I have been seeing the following problem for a while now:

During the transfer of the actual image Web Bluetooth is failing.

For example, sending a 24 kB image about half  the time everything will work. Otherwise usually I will see something in the log like:

> transferred: 600
> transferred: 1200
> ...
> transferred: 8400

Then we stall. Sometimes it ends early in the transfer, sometimes later. But Web Bluetooth hangs, doesn't print any error messages, and I see my remote BLE device disconnect and reset back into boot loader mode.

Just starting to really look into this issue now so details are vague, I will update the issue with my findings. (I need to test this more, its just a hunch right now, but I think this might be correlated to my environment. There are a lot of BLE devices in my office during the day. Call me crazy but it seems like I see this problem much more in the middle of the day than the end of the day - when their are less BLE devices communicating).

mjdietzx, 2016-03-17:
@beaufortfrancois: Have you seen anything like this?

thegecko, 2016-03-17:
This could be due to dropped packets?

mjdietzx, 2016-03-17:
That's what I'm wondering right now. You've never seen this problem before?

thegecko, 2016-03-17:
Yes, I have.

It's difficult to know whether this is a transport layer or protocol issue, however. Is there something we can listen to or timeout from to know to try resending s packet for example or is it that the underlying transport should guarantee delivery?

beaufortfrancois, 2016-03-18:
First time I see that. Let me have a look ;)

beaufortfrancois, 2016-03-18:
When it works fine, here's what I get in the end:

```
transferred: 21320
check length
length: 21548
validate...
complete, reset...
image activated and dfu target reset
disconnected and completed the DFU transfer
dfu complete
```

If it stalls though, here's what I get: http://pastebin.com/raw/bHbQuag4
As you can at the end, we get a disconnection which occurs for no apparent reason. Since `gattserverdisconnected` event is not triggered yet, there's no way of detecting this for now.

@vudentz, what do you think of this trace below?

1. Could the high number of Write Commands cause some issues?
2. Could it be a simple problem with Web Bluetooth only?

beaufortfrancois, 2016-03-18:
cc @scheib @jyasskin @g-ortuno

Vudentz, 2016-03-18:
@beaufortfrancois It looks like the device is being removed with pending writes still in progress causing the device to be disconnected:
@ Device Removed:...

Vudentz, 2016-03-18:
It seems after some point it switches from Write Command in handle 0x0e to Write Request in 0x11 and receives a Notification, perhaps that is used for confirming that everything has been received? But it looks like the device removed before that which seems wrong and perhaps that is the cause of the stall since the object is no longer available the request wont complete.

beaufortfrancois, 2016-03-18:
Could it be BlueZ removing the device from cache automatically?
The only time we remove a device is when a user goes to Chrome OS Settings and "forget" device manually.

g-ortuno, 2016-03-18:
@beaufortfrancois can you attach chrome logs?

beaufortfrancois, 2016-03-18:
I'm not sure what I'm doing different there but I can't reproduce it now... 
@mjdietzx can you now?

beaufortfrancois, 2016-03-18:
g-ortuno, could it be that `BluetoothGattConnectionBlueZ` gets destroyed because of memory issues?

```
BluetoothGattConnectionBlueZ::~BluetoothGattConnectionBlueZ() {
  bluez::BluezDBusManager::Get()->GetBluetoothDeviceClient()->RemoveObserver(
      this);
  Disconnect();
}
```

It looks like calling `Disconnect()` is the culprit here. Can you confirm?

beaufortfrancois, 2016-03-18:
After talking with @g-ortuno, we think you could hit a memory issue where reference to `gattServer` is garbage collected, hence terminating the connection.
@mjdietzx we suggest you set a global variable where to keep `gattServer` for now until this is fixed. 
Can you try this and let us know how that works out for you?

beaufortfrancois, 2016-03-18:
For info, I was able to reproduce the issue and gathered some logs at http://pastebin.com/raw/sbghuZHB
Reproducing it is easy when opening a tab at http://beta.unity3d.com/jonas/AngryBots/ and running the  the ASM.js fallback version of the game.

Vudentz, 2016-03-18:
@beaufortfrancois Actually the event @ Device Removed: actually refer to passive scanning, so it is indeed the device being disconnected.

mjdietzx, 2016-03-18:
Great, it seems like @beaufortfrancois solution to make gattserver a global variable fixed this problem.

beaufortfrancois, 2016-03-18:
YEAH!

beaufortfrancois, 2016-03-18:
@Vudentz I was confused because I thought "Device Removed" was the same as "remove 12:34:56:78:90:12" in bluetoothctl. Do you think it could be renamed then?

Vudentz, 2016-03-18:
@beaufortfrancois I was thinking the same, the problem is that the kernel management interface events have different meaning than D-Bus and its probably too late to change it.

thegecko, 2016-03-22:
Is there a bluez issue we can track for this?

g-ortuno, 2016-03-22:
There is nothing wrong with Bluez in this case. It's just the BluetoothDevice object getting garbage collected and disconnecting. As Francois mentioned a workaround is to keep a reference to the device or the gatt server. http://crbug.com/495232

beaufortfrancois, 2016-03-29:
@Vudentz I was more thinking about bluetoothd logs. Something like this maybe?

```diff
static void mgmt_device_removed(uint16_t len, const void *buf)
{
        const struct mgmt_ev_device_removed *ev = buf;
        char str[18];

        if (len < sizeof(*ev)) {
                printf("* Malformed Device Removed control\n");
                return;
        }

        ba2str(&ev->addr.bdaddr, str);

-        printf("@ Device Removed: %s (%d)\n", str, ev->addr.type);
+        printf("@ [Passive Scanning] Device Removed: %s (%d)\n", str, ev->addr.type);

        buf += sizeof(*ev);
        len -= sizeof(*ev);

        packet_hexdump(buf, len);
}
```

closed 2017-05-24 by thegecko