thegecko/web-bluetooth-dfu #19: Release?

thegecko, 2016-03-22:
@mjdietzx @beaufortfrancois 

I was thinking of pushing out a release to npm/bower.
Is there any features / bugfixes we want before this happens?

mjdietzx, 2016-03-22:
Do you think I should make a PR fixing this issue first?

https://github.com/thegecko/web-bluetooth-dfu/issues/18

Just basically making gattserver a global variable. Thats all i can think of for now.

jyasskin, 2016-03-24:
I'm really sorry to bring bad news, but I think we're going to have to blacklist the DFU service because even in v11 the default firmware doesn't support checking update signatures. The risk that people will get phished into malicious updates just seems too high.

https://github.com/WebBluetoothCG/registries/issues/7

thegecko, 2016-03-24:
sux.

jyasskin, 2016-03-24:
Very much, yes. :-(

beaufortfrancois, 2016-03-29:
Please let's start/continue this conversation at https://github.com/WebBluetoothCG/registries/issues/7

beaufortfrancois, 2016-04-08:
It has been applied to latest chromium build. See https://codereview.chromium.org/1866853002
@thegecko might want to update README.md for now. 

Don't worry though, we're still pursuing this with a potential signed DFU service ;)

thegecko, 2016-04-08:
On hols ATM, @beaufortfrancois feel free to create a PR

thegecko, 2017-05-22:
Published!

closed 2017-05-22 by thegecko