thegecko/web-bluetooth-dfu #22: Update to Secure DFU

beaufortfrancois, 2016-09-06:
Nordic just released a Secure version of their DFU at http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.0.0%2Fgroup__nrf__ble__dfu.html&cp=4_0_0_6_7_1_0 and I'd love to see web-bluetooth-dfu updated to support this.

thegecko, 2016-09-06:
That's great news, @mjdietzx is there still interest from Nordic to bring this library up to spec?

beaufortfrancois, 2016-10-03:
@mjdietzx @ketile might help there.

ketile, 2016-10-03:
I had a look at secure DFU this weekend to see what's new. In a rough estimate it seems doable in a week or two.  There are [Python](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/dfu_transport_ble.py) and  [Java](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/SecureDfuImpl.java) implementations which are helpful references. I'm putting together a flow diagram to visualize the process which I'll share once it is finished. This will hopefully help in further planning. I am still working on another project though so it will be a bit on/off from my side.

ketile, 2016-10-03:
Below are some of my notes which might be helpful.

# Web Bluetooth Secure DFU

* What has been done?
* What is new?
* What steps should be taken?
* General procedure
* Resources

### What has been done?
* [Secure DFU Java implementation](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/SecureDfuImpl.java)
* [Secure DFU Python implementation](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/dfu_transport_ble.py)
* [Legacy DFU JS implementation](https://github.com/thegecko/web-bluetooth-dfu)

### What is new?
* New init packet based on protocol buffers
* New Bootloader
* Manifest file for indicating package format
* CRC is now 32-bit. The old DFU CRC was 16-bit.
* Generic Bluetooth Object Transfer
* New UUIDs - DFU_SERVICE_UUID, DFU_CONTROL_POINT_UUID, DFU_PACKET_UUID
* ...

### What steps should be taken?
* Read through Java and Python implementations
* Read through legacy JS implementation
* Read through secure DFU documentation
* Draw flow diagram
* Code & Test

### General procedure
1. Connect to device. DFU Service must be a primary service.
 * 0xFE59
1. Enable Notifications for DFU Control Point Characteristic [...and wait a second?]( https://github.com/NordicSemiconductor/Android-DFU-Library/issues/10)
 * 0x8EC90001-F315-4F60-9FB8-838830DAEA50
1. Enable Notifications for the DFU Packet Characteristic.
 * 0x8EC90002-F315-4F60-9FB8-838830DAEA50
1. Set the Packet Receipt Notification (PRN) value. Optional. Default is 0.
1. Create Init packet
1. Calculate CRC32 and length of Init Packet
1. Send and verify Init packet
1. Send and verify Firmware

### Resources
* [BLE Secure DFU Transport documentation](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.0.0%2Flib_dfu_transport_ble.html&cp=4_0_0_3_3_3_2)

thegecko, 2016-10-03:
@ketile That's great news. In your opinion is this work which can extend the project we started here or something which should be considered from scratch?
I'll also be interested in how you plan approaching this. In the open? As part of this project or a new one? Whether you need any help?

I plan to update the [bleat](https://github.com/thegecko/bleat) library to match the current web-bluetooth specification over the next few weeks, so this could be a great test case for it.

ketile, 2016-10-03:
@thegecko I *think* it is different enough to warrant a new project. We can then borrow the parts that are reusable from your legacy implementation. I'll need to look into it more to se where they overlap. Otherwise this should be in the open and I'd love some help once we get going.

I'd prefer to host the project in a dedicated project folder in our [nrf5-physical-web repository]( https://github.com/NordicSemiconductor/nrf5-physical-web) with the usual Nordic Semiconductor license. That will hopefully make it easy for others to clone/fork and continue working on it.

What do you think?

ketile, 2016-10-03:
@thegecko On a second look it seems like most of the code that is not directly related to legacy DFU can be re-used. Some specific things that are new afaik:

* Change from 16-bit to 32-bit CRC
* It is currently not  possible to initiate DFU by writing to a characteristic
* New UUIDs for DFU service and characteristics
* New options for Control Point  and Data characteristic
* New init packet format

BTW I am not suggesting to change the license of your work. The important part is that it is not prohibiting our end users in any way :)

gfwilliams, 2016-12-13:
Hi - has anything progressed with this? With http://www.puck-js.com/ it'd be great to be able to add secure  updates into the Web IDE (which uses Web Bluetooth): https://www.espruino.com/ide

I can try and take a look at this, but if someone has already started it'd be better not to duplicate work :)

beaufortfrancois, 2016-12-29:
@ketile @thegecko Any update on this?

thegecko, 2016-12-29:
I've not been involved in the rewrite and can't see any additions to the folder mentioned by @ketile. @mjdietzx do you have any updates?

ketile, 2017-01-02:
I will be working on this but don't dare say when. It is high on the agenda though as there is quite a bit of interest.

OwenBrotherwood, 2017-02-14:
@ketile Any change of a time scale for first push to a repository ?

ketile, 2017-02-14:
@OwenBrotherwood Sorry about the delay. Priorities keep changing. Current goal is to have a working implementation by end of March.

urish, 2017-02-28:
I'd also love to see this happen :-) 
Let's cross fingers for end of March

urish, 2017-02-28:
I went ahead and create a PoC of Secure DFU with Web Bluetooth, also verified that it works with my nRF52. 

Code: https://github.com/urish/web-bluetooth-secure-dfu
Online version: https://urish.github.io/web-bluetooth-secure-dfu/

The core functionality is implemented as a plain ES2017 class, so it works out of the box on Chrome 55+ / Node 7.5.0, while older versions / different browsers require transpilation to either ES2015 or ES5. But no other browsers currently support Web Bluetooth, so it shouldn't be a big issue anyway.

Also, please forgive the quick-and-dirty UI.

ketile, 2017-03-01:
Thanks @urish ! Which DFU example did you work with? I was under the impression that the new Secure DFU [required a .zip file](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v13.0.0%2Fble_sdk_app_dfu_bootloader.html&cp=4_0_0_4_3_1) instead of separate init and firmware packets.

urish, 2017-03-01:
@ketile if you extract the zip file, you will find two files inside: a .bin and .dat file, corresponding the two forms field.

urish, 2017-03-01:
I worked with the DFU code from Espruino: https://github.com/espruino/Espruino/tree/0b0a032ff80a7f9d162d27a72097dc7d02324cc2/targets/nrf5x_dfu

ketile, 2017-03-01:
@urish Great! Will test it right away.

ketile, 2017-03-01:
Works great @urish

beaufortfrancois, 2017-04-27:
Do you folks have any plan of using @urish's code?

gfwilliams, 2017-04-27:
I will at some point - I just have to get around to integrating it into the Espruino Web IDE

beaufortfrancois, 2017-04-27:
That would be great !
Please let us know when you have time ;)

ketile, 2017-04-27:
We are already using @urish's secure DFU solution as a library in a yet to be disclosed application :)

thegecko, 2017-04-27:
Ooh, secrets :)

What are people's thoughts on merging this repo with @urish or deprecating this?

urish, 2017-04-29:
Ping me if you need any assistance

thegecko, 2017-05-14:
See #24

thegecko, 2017-05-22:
Published!

closed 2017-05-22 by thegecko