thegecko/web-bluetooth-dfu #23: Similar DFU mechanism using bleat Classic API

phearme, 2017-05-08:
Hey!
I am trying to perform a similar DFU mechanism in nodeJS using bleat classic API instead of the web bluetooth one. I seem to have everything hooked up, but can't get any callback on notifications from the control characteristic, furthermore I cannot find any good example on how to use the enableNotify correctly. I can write OPCODEs to the control characteristic, but the notification callback seems to be never called. Thanks for any clue on this.

phearme, 2017-05-08:
never mind. it seems we have the same issue using also the web bluetooth api.

closed 2017-05-08 by phearme

thegecko, 2017-05-08:
Keep an eye on this:

https://github.com/thegecko/web-bluetooth-dfu/tree/version-13

phearme, 2017-05-08:
@thegecko thanks for your answer, I am using nrg52832 sdk 11 sd s132. Version 2.0. I tried the version-13 branch but when I select my device I get `noble warning: unknown peripheral xxxxx`. furthermore, according to my hardware guys, the target device I am using does not support secure dfu (at least according to them).

using the previous method (master branch), here is the output:
```
found device: DfuTarg
using file name: firmware/nrf52_app.hex
connected to device
found DFU service
found control characteristic
found packet characteristic
found version characteristic
enabling notifications
writing modeData
modeData written
DFU target issued GAP disconnect and reset into bootloader/DFU mode
start address of binary: 110592
end address of binary: 135144
ignore records of type start linear address
done converting hex file to binary
file length: 24552
connected to device
found DFU service
found control characteristic
found packet characteristic
found version characteristic
read versionChar
using dfu version 8.0
enabling notifications
sending imagetype: 4
sent start
sent image size: 24552
disconnected and completed the DFU transfer
```
setting write mode seems to work, the transfer also, but I don't see any events coming in, don't know if this normal, but the device doesn't seem to have updated anything.
also, we are using custom UUIDs for DFU, notice 1540 instead of 1530 :
```
    var serviceUUID = "00001540-1212-efde-1523-785feabcd123";
    var controlUUID = "00001541-1212-efde-1523-785feabcd123";
    var packetUUID = "00001542-1212-efde-1523-785feabcd123";
    var versionUUID = "00001544-1212-efde-1523-785feabcd123";
```

thegecko, 2017-05-08:
The current code in master targets version `v0.9.2` of the DFU, so I'm not sure how well it will work with version 11.
Secure DFU was introduced in version 12 :(

phearme, 2017-05-08:
yeah I figured also. strange thing is we are able to do DFU successfully using the Android nRF Toolbox app, so I guess it should also be possible using bleat. the only thing I cannot understand is why are we not receiving any notification after sending `OPCODE.START_DFU` with the imageType.

thegecko, 2017-05-08:
The app probably supports multiple versions of the protocol, whereas this repo has stopped at 0.9.2 and will (currently) start again from v13. It's very possible the protocol has changed.

Compare this:

http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk52.v0.9.2%2Fbledfu_transport.html&cp=4_0_2_4_2_4

With this:

http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v11.0.0/bledfu_transport.html

Any differences would needed to be catered for.

phearme, 2017-05-08:
@thegecko thanks for your help.

thegecko, 2017-05-08:
No problem. Let me know what you find.

phearme, 2017-05-08:
Both protocol specs are identical. I am wondering if using custom UUIDs for the service, control characteristic and packet characteristic has an impact. will revert them to default and see if that changes anything.

phearme, 2017-05-08:
UPDATE: using default UUIDs has no impact. it seems that bleat isn't receiving any notification whatsoever even though notify is supported on the control characteristic.

thegecko, 2017-05-21:
Did you resolve this issue @phearme ?

phearme, 2017-05-22:
@thegecko hi. no, never got to solve it. here are my investigation on it:
I am able to do DFU over ble through the Android app, and I noticed the LegacyDFU was used which is normal since we are using SDK 11.
I tried to use also the nrfutil command line tool: the current version of nrfutil only supports SDK 12 and above for DFU over ble. they do have a version 0_5_2 which supports old versions of SDK but that one doesn't do DFU over ble, it only does DFU over serial.
I have tried the bleat library for nodejs both with classic api and with web bluetooth api, and with both I have the same behavior: I am able to connect, set the device in write mode, but am not receiving any notification whatsoever.

thanks for any clue at all on this. I really want to get this working.

thegecko, 2017-05-22:
What's your setup?
 - hardware
 - softdevice
 - stock bootloader?
 - node version?

phearme, 2017-05-22:
node version: v6.10.3
stock bootloader: from sdk 11 examples
soft device: 2.0
hardware: custom hardware based on nRF52832

thegecko, 2017-05-25:
I've just confirmed the legacy DFU update works with the setup you have (or as close as!)

platform: MacOSX
hardware: [nRF52-DK](http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK)
SDK: [version 11](https://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v11.x.x/nRF5_SDK_11.0.0_89a8197.zip)
softdevice: [2.0.1](http://www.nordicsemi.com/eng/nordic/download_resource/51481/6/55708700/95151)
bootloader: [SDK]/examples/dfu/bootloader/hex/dfu_dual_bank_ble_s132_pca10040.hex
app: [SDK]/examples/dfu/ble_dfu_send_hex/test_images_update_nrf52/dfu_test_app_hrm_s132.hex
node versions tested: 4.7, 6.10, 7.10

I'm happy to send you the test project if it helps?

phearme, 2017-05-25:
@thegecko thanks so much for your help. if the test project is identical to the current branch on github, no need to. maybe something on our hardware side.

thegecko, 2017-05-25:
[phearme.zip](https://github.com/thegecko/web-bluetooth-dfu/files/1028778/phearme.zip)

phearme, 2017-05-25:
@thegecko thanks dude. I'll give it a try and tell you how it went.

phearme, 2017-05-25:
@thegecko I don't see any difference with the method we are using on our side. I was sure there was something wrong with the hardware because we don't get any notifications, and I'll investigate on that side. maybe the bluetooth embedded on my pc sucks, and I should use an official nordic dongle.

thegecko, 2017-05-25:
Which OS are you running? It may be a problem in the [noble](https://github.com/sandeepmistry/noble) stack below bleat?

phearme, 2017-05-26:
@thegecko using Ubuntu 16.04
Didn't try on another OS, good point. I will try on a Mac and let you know.