thegecko/web-bluetooth-dfu #25: Not working using node.js on Windows

lemmerelassal, 2017-09-03:
Hi there,

I've been trying to use your web interface to update Nordic nRF52832 using node.js, but always end up with this:

 
Select a device to update:
1: DfuTarg
2: null
3: null
4: null
5: null
6: null
7: null
8: null
9: null
10: null
11: null
 
DfuTarg selected, connecting...
DFU mode set
requestDevice error: no devices found

Your web interface works from my Android phone (Samsung S4) but not using Chrome nor Firefox on Windows.

Can you please help me?

lemmerelassal, 2017-09-03:
I narrowed it down to

secure-dfu.js:

```
    secureDfu.prototype.update = function(device, init, firmware) {
        if (!device) throw new Error("Device not specified");
        if (!init) throw new Error("Init not specified");
        if (!firmware) throw new Error("Firmware not specified");

        return this.connect(device)
        .then(() => {
            console.log("transferring init");
            return this.transferInit(init);
        })
```

Without a breakpoint at console.log("transferring init"), it won't print that line in console.
This is on Windows 10.

Thanks in advance.

lemmerelassal, 2017-09-03:
Narrowed it down some more:

After this:
```
    secureDfu.prototype.transfer = function(buffer, type, selectType, createType) {
        return this.sendControl(selectType)
```
... it won't enter: 
`        .then(response => {`

thegecko, 2017-09-03:
With regards to browser support, I don't believe firefox supports web bluetooth at all. Checking the implementation table for chrome (https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md) it isn't completely implemented in Chrome on Windows, either.

The node implementation was tested on MacOS (I don't have a windows machine), but the bluetooth layer is abstracted using noble (https://github.com/sandeepmistry/noble) and should be the same across all OSs.

I recommend the next step is to try some of the noble examples to confirm you can connect to a bluetooth device from your machine.

Debugging may also be easier using a tool such as node-inspector (https://github.com/node-inspector/node-inspector)

lemmerelassal, 2017-09-04:
Thank you... I do use node-inspector. That's how I figured out the exact line it has problems with. Same confirmed using VS Code.
I can actually detect the device, select it, but it hangs when sending control.

lemmerelassal, 2017-09-04:
I tried on Ubuntu 16.04 LTS, Chrome 60, and it lists the devices (unlike on Windows 10), but still won't proceed.

Any clue what to do?

thegecko, 2018-01-08:
@lemmerelassal Sorry it's been a while getting back to you. The library has just been rewritten in TypeScript and a few issues were resolved along the way. I'd appreciate it if you can confirm you still see an issue with the latest version. If you do can you outline:

- The OS you are using
- Whether you are updating use a web page or Node.js
- The device being used
- The Soft Device in use

thegecko, 2018-02-27:
Closing due to inactivity...

Noggin01, 2018-03-08:
Can we reopen this issue?  I'm seeing a similar issue using a BeagleBone Green Wireless (Debian 9.3) as well as Ubuntu 16.04 on a PC.

I can bootload my board using my Android phone (Pixel XL, 8.1.0) and a web browser as well as using Nordic's application nRF Connect.  My board is the Nordic development board, PCA10040.  The bootloader is their example bootloader and is build using SDK 12.0.0.  I have not modified the bootloader in any way, except to prove my own security key for the bootload file.

I am unable to bootload via the following methods:

- Chrome Beta in Windows 7.  The system always indicates that Bluetooth LE is not available.
- Chrome Beta in Ubuntu 16.04.  The system will pair to the device, but doesn't seem to ever initiate the transfer.  
- Bash on the BeagleBone Green Wireless (BBGW) (more below)  
- Bash on Ubuntu 16.04 (more below)

My PC, which runs Windows and Ubuntu (in an Oracle Box virtual machine), does not have built in Bluetooth, so I'm using a USB dongle with a BCM20702A0 chipset.

Ultimately, I need to be able to bootload via the BBGW.  I seem to be having the same experience on both Ubuntu the BBGW.

I've edited your example node.js file as follows:

```
function update() { 
   var package = null;
   var device = null;
   var dfu = null;

   // Create dfu here so we can add the event listener
   dfu = new SecureDfu(crc.buf, webbluetooth.bluetooth);
   dfu.addEventListener(SecureDfu.EVENT_LOG, event => {
        console.log("Event:", event.message)
   })

   // everything below this line is the same, except that I have commented out the dfu = new SecureDfu(...)
```

Here's my output: 
```
Enter a URL or file path for the firmware package: myDFU.zip
Scanning for DFU devices...
Select a device to update:
1: DfuTarg
2: Anova
3: Unknown or Unsupported Device (e9:8b:ec:b1:a9:0f)
4: Mooshimeter V.1
5: One
DfuTarg selected, connecting...
Event: connected to gatt server
Event: found DFU service
Event: found 2 characteristic(s)
Using firmware: SW-0511-8.bin
Event: connected to gatt server
Event: found DFU service
Event: found 2 characteristic(s)
Event: found packet characteristic
Event: found control characteristic
Event: enabled control notifications
Event: transferring init
```

It looks to me that `private sendOperation(...)` is never resolving.  From what I can tell, `characteristic.writeValue(value);` is getting called.

Noggin01, 2018-03-09:
I now have an application based on SDK 14.3.0 which has Buttonless DFU implemented.  I've attempted to get a bootload working by invoking it through the application, but still no luck.

```
Scanning for DFU devices...
Select a device to update:
1: Unknown or Unsupported Device (7b:56:f0:9c:bc:9e)
2: Anova
3: Nordic_Buttonless
4: Mooshimeter V.1
5: Unknown or Unsupported Device (98:d6:bb:0c:d4:21)
6: Unknown or Unsupported Device (d1:39:12:05:17:ba)
7: Unknown or Unsupported Device (ec:ee:38:fd:f6:19)
8: One
9: Unknown or Unsupported Device (ec:fe:7e:10:9c:50)
Nordic_Buttonless selected, connecting...
Event: connected to gatt server
Event: found DFU service
Event: found 1 characteristic(s)
Event: found buttonless characteristic
Event: enabled buttonless notifications
notifyFns: {}
Operation: [ 1 ]
notifyFns: { '1': { resolve: [Function], reject: [Function] } }
```

Noggin01, 2018-03-10:
I think I'm narrowing in on the issue.  I have an nrf51 dongle I was able to configure as a BLE sniffer.  Using it, I captured a successful bootload from my phone vs a failed bootload from node.  What I saw is that when the node bootload fails, it sends the [6, 1] as a WriteCommand whereas when my phone sends it, it is sent as a WriteRequest.

WriteRequest expects a response where as a WriteCommand is a write-and-forget.  I think the response is critical in the bootload procedure, so this needs to be changed to a WriteRequest.  I'm going to start looking into how to do that, but I must admit I'm a C programmer.  Node is little difficult for me, but I'm persistent.

I ran the node code in node-inspector and set a breakpoint at the start of Characteristic.prototype.write, line 62.

```
Characteristic.prototype.write = function(data, withoutResponse, callback) {
  if (process.title !== 'browser') {
    if (!(data instanceof Buffer)) {
      throw new Error('data must be a Buffer');
    }
  }

  if (callback) {
    this.once('write', function() {
      callback(null);
    });
  }

  this._noble.write(
    this._peripheralId,
    this._serviceUuid,
    this.uuid,
    data,
    withoutResponse
  );
};
```
I set withoutResponse to false every time I hit the breakpoint, and the bootload proceeded as desired, though a little slow.

```
Select a device to update:
1: DfuTarg
2: Anova
3: Unknown or Unsupported Device (ec:fe:7e:10:9c:50)
4: Unknown or Unsupported Device (98:d6:bb:0c:d4:21)
5: Unknown or Unsupported Device (7a:c0:e7:67:e2:81)
6: One
7: Mooshimeter V.1
8: Unknown or Unsupported Device (f1:9d:d2:68:9a:45)
9: Unknown or Unsupported Device (e6:3f:b8:77:0a:21)
DfuTarg selected, connecting...
Event: connected to gatt server
Event: found DFU service
Event: found 2 characteristic(s)
Using firmware: SW-0562-x.x.bin
Event: connected to gatt server
Event: found DFU service
Event: found 2 characteristic(s)
Event: found packet characteristic
Event: found control characteristic
Event: enabled control notifications
Event: transferring init
Transfer: [ 6, 1 ]
notifyFns: {}
Operation: [ 6, 1 ]
notifyFns: { '6': { resolve: [Function], reject: [Function] } }
Write Value: Uint8Array [ 6, 1 ]
Event: handle notification
Event: Response
DataView {
  byteLength: 12,
  byteOffset: 3,
  buffer: ArrayBuffer { byteLength: 15 } }
Event: { object: 'init', totalBytes: 135, currentBytes: 0 }
notifyFns: {}
Operation: [ 1, 1 ]
notifyFns: { '1': { resolve: [Function], reject: [Function] } }
Write Value: Uint8Array [ 1, 1, 135, 0, 0, 0 ]
Event: handle notification
Event: { object: 'init', totalBytes: 135, currentBytes: 20 }
Event: { object: 'init', totalBytes: 135, currentBytes: 40 }
Event: { object: 'init', totalBytes: 135, currentBytes: 60 }
Event: { object: 'init', totalBytes: 135, currentBytes: 80 }
Event: { object: 'init', totalBytes: 135, currentBytes: 100 }
Event: { object: 'init', totalBytes: 135, currentBytes: 120 }
Event: { object: 'init', totalBytes: 135, currentBytes: 135 }
notifyFns: {}
Operation: [ 3 ]
notifyFns: { '3': { resolve: [Function], reject: [Function] } }
Write Value: Uint8Array [ 3 ]
Event: handle notification
Event: written 135 bytes
notifyFns: {}
Operation: [ 4 ]
notifyFns: { '4': { resolve: [Function], reject: [Function] } }
Write Value: Uint8Array [ 4 ]
Event: handle notification
Event: transfer complete
Event: transferring firmware
Transfer: [ 6, 2 ]
notifyFns: {}
Operation: [ 6, 2 ]
notifyFns: { '6': { resolve: [Function], reject: [Function] } }
Write Value: Uint8Array [ 6, 2 ]
Event: handle notification
Event: Response
DataView {
  byteLength: 12,
  byteOffset: 3,
  buffer: ArrayBuffer { byteLength: 15 } }
notifyFns: {}
Operation: [ 1, 2 ]
notifyFns: { '1': { resolve: [Function], reject: [Function] } }
Write Value: Uint8Array [ 1, 2, 0, 16, 0, 0 ]
Updating application [                    ] 0% 0.0sEvent: handle notification
Updating application [                    ] 0% 27506.1s
```

thegecko, 2018-03-10:
You certainly are persistent! Sorry I don't have time for a few weeks to support you, but good luck. Sounds like you are onto something.

Noggin01, 2018-03-10:
Bootload is successful!

I went into node_modules/webbluetooth/adapter.js and changed
`this.characteristicHandles[handle].write(buffer, true, this.checkForError(errorFn, completeFn));`
to 
`this.characteristicHandles[handle].write(buffer, false, this.checkForError(errorFn, completeFn));`

I was able to bootload my device, put it back into DFU mode via buttonless DFU, and bootload it again.  I did have to restart the script after putting it back into DFU mode, but I can deal with that for now.

Now to try and replicate this on my BeagleBone....

thegecko, 2018-03-11:
Good work!

Does this PR fix this issue for you?
https://github.com/thegecko/webbluetooth/pull/4

Noggin01, 2018-03-12:
Unfortunately, it does not appear to have resolved the issue.  I'm not 100% certain I did this correctly though.  Not only am I a C programmer, but I'm also an SVN user.  My steps were:

1. Verified my bootload was working
2. cd node_modules
3. rm webbluetooth -rf
4. git clone https://github.com/thegecko/webbluetooth.git
5. cd webbluetooth
6. git checkout b/write-detect
7. npm install
8. npm run gulp
9. nano ./lib/adapter.js // to verify that the updated writeCharacteristic is in place

The bootload progress bar immediately indicates 5% completed and never proceeds.

Noggin01, 2018-03-12:
Wait!  It worked if I changed

`const withoutResponse = characteristic.properties.indexOf("writeWithoutResponse") >= 0;`

to 

`const withoutResponse = characteristic.properties.indexOf("writeWithoutResponse") > 0;`

thegecko, 2018-03-12:
Your setup correctly incorporates the changes.
I was hoping the added code would result in false being passed, but I think your test proves that the device advertises it can do a write without response, but doesn't handle it somewhere down the stack. I can switch the change to default to writing with a response if that's detected, which should fix your issue.

thegecko, 2018-03-13:
Could you confirm this by adding a log outputting the properties exposed by the device? e.g.:

```javascript
// tslint:disable-next-line:no-console
console.log(characteristic.properties);
const withoutResponse = characteristic.properties.indexOf("writeWithoutResponse") >= 0;
```

thegecko, 2018-03-13:
Could you try the same branch again, now?

Noggin01, 2018-03-13:
I removed all of my extraneous console logging and tried the branch.  It jumps to 5% and immediately stops.  Normally, when the bootload process is working, it has to work its way up to 5%.

With the additional logging you requested above, this is my output:

```
Scanning for DFU devices...
Select a device to update:
1: Nordic_Buttonless
2: Unknown or Unsupported Device (c5:64:ba:5d:39:cd)
3: Mooshimeter V.1
4: Unknown or Unsupported Device (d9:67:c6:3c:cd:37)
5: Anova
6: Unknown or Unsupported Device (98:d6:bb:0c:d4:21)
7: Unknown or Unsupported Device (e6:3f:b8:77:0a:21)
8: One
Nordic_Buttonless selected, connecting...
[ 'write', 'indicate' ]
(node:2803) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'notifyFns' of undefined
    at complete (/home/henrymcdonald/dfu/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:386:36)
    at /home/henrymcdonald/dfu/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:399:21
    at <anonymous>
(node:2803) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:2803) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
DFU mode set, reconnecting...
Using firmware: SW-0562-x.x.bin
[ 'write', 'notify' ]
[ 'write', 'notify' ]
[ 'write', 'notify' ]
Updating application [                    ] 0% 0.0s[ 'writeWithoutResponse' ]
[ 'writeWithoutResponse' ]
[ 'writeWithoutResponse' ]
[ 'writeWithoutResponse' ]
[ 'writeWithoutResponse' ]  // I deleted 200 instances of ['writeWithoutResponse'] from the log
[ 'write', 'notify' ]
Updating application [=                   ] 5% 0.9s
```

Similar run, but I added the value of withResponse:

```
Scanning for DFU devices...
Select a device to update:
1: Unknown or Unsupported Device (98:d6:bb:0c:d4:21)
2: Anova
3: Mooshimeter V.1
4: Unknown or Unsupported Device (ec:fe:7e:10:9c:50)
5: One
6: Nordic_Buttonless
7: Unknown or Unsupported Device (d9:67:c6:3c:cd:37)
Nordic_Buttonless selected, connecting...
withResponse: true Properties: [ 'write', 'indicate' ]
(node:2975) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'notifyFns' of undefined
    at complete (/home/henrymcdonald/dfu/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:386:36)
    at /home/henrymcdonald/dfu/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:399:21
    at <anonymous>
(node:2975) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:2975) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
DFU mode set, reconnecting...
Using firmware: SW-0562-x.x.bin
withResponse: true Properties: [ 'write', 'notify' ]
withResponse: true Properties: [ 'write', 'notify' ]
withResponse: true Properties: [ 'write', 'notify' ]
Updating application [                    ] 0% 0.0swithResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]  // 200 instances of this line were removed
withResponse: true Properties: [ 'write', 'notify' ]
Updating application [=                   ] 5% 0.8s

```

thegecko, 2018-03-13:
This is really odd, the program needs to write to two different chars. The first of which only allows standard writing *with* a response, so that gets set correctly:

- withResponse: true Properties: [ 'write', 'indicate' ]

The second only accepts a write *without* a response and that is also correct:

- withResponse: false Properties: [ 'writeWithoutResponse' ]

I wonder if making your system wait for a response in the last case (which it won't get) actually introduces a delay which avoids a race condition somehow on Windows.

This could be tested by introducing an (async) delay before returning. e.g. something like:

``` javascript
        characteristic.write(buffer, true, function(err) {
            setTimeout(completeFn, 500);
        });
```

Noggin01, 2018-03-13:
Just to be clear, I'm not convinced that this is a Windows issue.  I'm running in a Virtual Machine, which to my understanding, virtually bypasses all of Windows controls and lets Ubuntu run just about down on the bare metal.  I'm also experiencing the exact same issues running on my BBGW with Debian.

However, I do agree that something odd seems to be at play.

Here's some more logging output.  The main difference is that I've enabled event logging from your event emitter in secureDfu.ts:

```
Scanning for DFU devices...
Select a device to update:
1: DfuTarg
2: Anova
3: Unknown or Unsupported Device (98:d6:bb:0c:d4:21)
4: One
DfuTarg selected, connecting...
Event: connected to gatt server
Event: found DFU service
Event: found 2 characteristic(s)
Using firmware: SW-0562-x.x.bin
Event: connected to gatt server
Event: found DFU service
Event: found 2 characteristic(s)
Event: found packet characteristic
Event: found control characteristic
Event: enabled control notifications
Event: transferring init
withResponse: true Properties: [ 'write', 'notify' ]
Event: handle notification
Event: Response
Event: { object: 'init', totalBytes: 135, currentBytes: 0 }
withResponse: true Properties: [ 'write', 'notify' ]
Event: handle notification
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 20 }
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 40 }
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 60 }
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 80 }
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 100 }
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 120 }
withResponse: false Properties: [ 'writeWithoutResponse' ]
Event: { object: 'init', totalBytes: 135, currentBytes: 135 }
withResponse: true Properties: [ 'write', 'notify' ]
Event: handle notification
Event: written 135 bytes
withResponse: true Properties: [ 'write', 'notify' ]
Event: handle notification
Event: transfer complete
Event: transferring firmware
withResponse: true Properties: [ 'write', 'notify' ]
Event: handle notification
Event: Response
withResponse: true Properties: [ 'write', 'notify' ]
Updating application [                    ] 0% 0.0sEvent: handle notification
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: false Properties: [ 'writeWithoutResponse' ]
withResponse: true Properties: [ 'write', 'notify' ]
Updating application [=                   ] 5% 0.4s
```
I'm too much of a C programmer to get your setTimeout work just as you described (kept getting errors about either errorFn or completeFn not being called, or not allowing them to be set to void, etc) but I did get something kludged in there and it seems to have worked.  

```
    public writeCharacteristic(handle: string, value: DataView, completeFn?: () => void, errorFn?: (errorMsg: string) => void): void {
        const buffer = this.dataViewToBuffer(value);
        const characteristic = this.characteristicHandles[handle];

        // Default to writing with a response if supported (writeWithoutResponse and authenticatedSignedWrites don't require a response)
        const withResponse = characteristic.properties.indexOf("write") >= 0
                          || characteristic.properties.indexOf("reliableWrite") >= 0;

        // tslint:disable-next-line:no-console
        console.log("withResponse:", withResponse, "Properties:", characteristic.properties);

        if (withResponse) {
                characteristic.write(buffer, !withResponse, this.checkForError(errorFn, completeFn));
        } else {
                characteristic.write(buffer, !withResponse, function(){ setTimeout(completeFn, 20);});
        }
    }
```

In looking at the output, while the firmware data is being transferred to the Nordic board, `withResponse` is false.  If I understand this correctly, this means that we're writing data to the DFU and not waiting for a response, but immediately sending data on the next process tick (or as soon as Node can do it).  This doesn't seem reasonable to me.  I'm not saying that your interpretation of the DFU process is wrong, but I would imagine that we _need_ delays in place to give the MCU time to write to flash.  Writing to flash takes an eternity in the timescales of a microcontroller.

One of the things that I noticed was different about the Nordic Android application and your example is that Android application changes the connection parameters.  I didn't pay this much heed, but in the log I collected it appears that the connection interval was changed to 45 ms.  I wonder if that means that the the characteristics can only be written at 45mS intervals.

thegecko, 2018-03-13:
OK, I'll make some more changes later and get you to test them if you don't mind ;)
Thanks for being a guinea-pig!

thegecko, 2018-03-16:
@Noggin01 If you have a spare few minutes, can you test the changes on my branch again?
I've introduced a delay, but dropped it right down. I think it may still work and be quicker, but your help testing it would be useful. If it doesn't work, could you perhaps find the lowest number where it does? 0, 1, 5, 10, 20, 50, 100, etc. Thanks!

Noggin01, 2018-03-16:
50 - Successful, bootload time 224.72 seconds
25 - Successful, bootload time 125.1 seconds
12 - Successful, 60.01 seconds
6 - Failed at 10% through bootloading
10 - Failed at 20% through bootloading
11 - Successful, bootload time 56.7 seconds

The timings above did NOT include the time required to switch from application mode to DFU.  It only included the time required to send the file after connecting to the device.

Based on the above testing, 10 is obviously not long enough of a delay to achieve a successful DFU.  I would suggest no less than 25.

thegecko, 2018-03-16:
Awesome, thanks!

PR #27 (2018-03-17): Minor fixes
- Fix tslint build
- Disable deletion when watching
- Updated webbluetooth library, fixes #25
closed 2018-03-17 by thegecko