thegecko/web-bluetooth-dfu #28: Error: "GATT operation failed for unknown reasons"

fhaubold, 2018-07-17:
Hi,

unfortunately, I can't get your module working. Even your online demo does not work.

#### Steps to reproduce
1. Select firmware package
2. Select device (nrf52)
3. Reselect device (DfuTarg)
4. Copying starts, but after a few seconds the error appears

#### Setup
macOS 10.13.6
Chrome 67
Buttonless nrf52 device

#### Console output
```
web.html:214 connected to gatt server
web.html:214 found DFU service
web.html:214 found 1 characteristic(s)
web.html:214 found buttonless characteristic
web.html:214 enabled buttonless notifications
web.html:214 sent DFU mode
secure-dfu.js:448 Uncaught (in promise) TypeError: Cannot set property 'notifyFns' of undefined
    at n (secure-dfu.js:448)
    at secure-dfu.js:461
n @ secure-dfu.js:448
(anonymous) @ secure-dfu.js:461
Promise.then (async)
(anonymous) @ secure-dfu.js:459
(anonymous) @ secure-dfu.js:446
Promise.then (async)
c.setDfuMode @ secure-dfu.js:424
(anonymous) @ secure-dfu.js:411
Promise.then (async)
c.requestDevice @ secure-dfu.js:409
selectDevice @ web.html:220
web.html:214 connected to gatt server
web.html:214 found DFU service
web.html:214 found 2 characteristic(s)
web.html:214 connected to gatt server
web.html:214 found DFU service
web.html:214 found 2 characteristic(s)
web.html:214 found packet characteristic
web.html:214 found control characteristic
web.html:214 enabled control notifications
web.html:214 transferring init
web.html:214 init packet already available, skipping transfer
web.html:214 transferring firmware
web.html:1 Uncaught (in promise) DOMException: GATT operation failed for unknown reason.
```

Thank you!

thegecko, 2018-07-17:
Which board are you using and what version of the soft device?

fhaubold, 2018-07-20:
I have checked the Chrome Bluetooth logs, the connection timed out:

```
<GATT disconnected, "DfuTarg">: Bluetooth error: error domain: CBErrorDomain, code: 6, description: The connection has timed out unexpectedly.
```

Why does this happen?

fhaubold, 2018-07-23:
I think I finally found the problem. The write operations are too fast for the device. At some point, packages get lost and the final call to calculate the CRC checksum also get lost which causes the weird error message as no response comes in.

I think the best solution would be to change the  BLE connection interval of the device, however, as a quick hack, a delay of 10 ms after each write (see below) works very well for me. Of course, this makes the DFU process very slow.

#### Quick solution ####
I added the following helper function to `secure-dfu.ts`:
```
private delayPromise(delay) {
  return data => {
    return new Promise((resolve, _) => {
      setTimeout(() => {
        resolve(data);
      }, delay);
    });
  };
}
```
Then I used it in the function `transferData()`:
```
...
return this.packetChar.writeValue(packet)
  .then(this.delayPromise(10))
  .then(() => {
...
```
I hope this helps someone!

closed 2018-07-23 by fhaubold

thegecko, 2018-07-23:
I'm glad you found the problem. I've seen issues like this before even with `WriteWithResponse` based characteristics.

Feel free to turn your fix into a PR where the delay can be passed into the SecureDFU constructor. I'd also be intrigued to know of your fix still works with a delay of 0 (effectively resetting the JavaScript stack).

fhaubold, 2018-07-24:
Done! With a delay of 0 the error appears again. Even with 5, it happens sometimes ...