thegecko/web-bluetooth-dfu #30: Unable to select a device on Windows 10 Chrome (Canary)

louisn, 2018-09-14:
A bluetooth device cannot be selected when running on Windows 10 Chrome (Canary) with the chrome://flags/#enable-experimental-web-platform-features enabled.

secure-dfu.js line 417 is reporting 'Cannot read property requestDevice of undefined.

thegecko, 2018-09-14:
That error sounds like `navigator.usb` is not available which suggests web-bluetooth hasn't been enabled.

louisn, 2018-09-14:
It has, I am connecting and interacting with a device using another
application on the same computer. The DFU feature doesn't appear to work.

louisn, 2018-09-24:
I haven't resolved any issue with this app.  However I was able to get the application I'm working to select a device.

closed 2018-09-24 by louisn