thegecko/web-bluetooth-dfu #33: Instructions for Arduino users

dsteinman, 2019-06-22:
Hi, I'd like to try this program out, but I am not using Nordic's SDK.  I wrote my firmware with Arduino for the Adafruit nRF52840, and I can export a .hex file.  But that's about where I'm stuck and not sure what the next steps are to create a DFU package that's properly signed.

- Rebuild the bootloader with your new key (Update the Makefile to use your new key file)

I have a JTAG Segger device and can flash the Adafruit nRF52 bootloader hex, but I don't know what to change in the Bootloader so that it's signed.

https://github.com/adafruit/Adafruit_nRF52_Bootloader

I'm having trouble figuring out if this is even possible to do with Arduino and Adafruit's bootloader.

If you know of any resources I can look at it would be very much appreciated.

thegecko, 2019-06-22:
Sorry, I've not used that device so can't comment whether it's possible.

dsteinman, 2019-06-25:
I looked into a bit more, and unfortunately it looks like it's not possible (at least not yet) because Adafruit's bootloader uses the older legacy DFU mode (not secure).  More info here https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/63

I was only able to get OTA updates with their Bluefruit mobile app.

closed 2019-06-25 by dsteinman