thegecko/web-bluetooth-dfu #34: How does Web Blueotth DFU work ?

il3ven, 2019-09-01:
I am trying to develop similar DFU for Nordic but using WebUSB. So I was first trying to understand Web Bluetooth.

So far I have understood this much -
- Bootloader is unchanged
- Secure DFU class has functions for transferring, updating

Can anybody give an overview of how different components work together in this code ? Components such as Web Bluetooth, Nordic's Bootloader, SoftDevice.

thegecko, 2019-09-01:
The protocols and approach used are all outlined in Nordic's SDK documentation as linked in the README

closed 2019-09-01 by thegecko

il3ven, 2019-09-01:
Okay. Nordic's SDK define protocols for DFU upgradation. Is it right to summarise that this project uses the same protocol as defined in SDK but instead using Bluetooth from within an app this uses from within chrome using Web Bluetooth API ?

thegecko, 2019-09-01:
Yes