thegecko/web-bluetooth-dfu #38: Uncaught (in promise) DOMException: GATT operation already in progress.

hugobfr, 2020-04-01:
Hello there !

I am using your tool for my personal use, it is great and easy to adjust to my needs.

I just want to open an issue for anyone who might stumble upon this unhandled exception as I have found a workaround.

This exception can appear randomly during a firmware update:
`Uncaught (in promise) DOMException: GATT operation already in progress.`
The update stops, and needs to be restarted in order to resume the transfer.

After some investigation, I found out this exception is raised by the characteristic write in the function `sendOperation`.

The workaround I found is the following:

```
private sendOperation(characteristic: BluetoothRemoteGATTCharacteristic, operation: Array<number>, buffer?: ArrayBuffer): Promise<DataView> {
             ...
            characteristic.writeValue(value)
            .catch(() =>  {
                this.log("DOMException: GATT operation already in progress.")
                return Promise.resolve()
                    .then(() => this.delayPromise(500))
                    .then(() => { characteristic.writeValue(value);});
            });
```

Now, everyting works fine and the update continues when the exception is catched.

[image]

thegecko, 2020-04-01:
Thanks! Can you open a PR to discuss?

jins-tkomoda, 2020-05-09:
I appreciate your great job! As I encountered the same issue frequently, I make PR instead of @hugobfr.

commit 4f640989 (2020-05-10): Fix #38 issue
commit d050b894 (2020-05-10): Merge pull request #42 from jins-tkomoda/pr-i38

Fix #38 issue
closed 2020-05-10 by thegecko