thegecko/web-bluetooth-dfu #39: Unable to find DFU service

pacmac, 2020-04-19:
I am using debian 9, and attempted to run the node.js example.

It successfully lists my nearby device, and when I select the correct device, it errors with the following:

```
connecting...
(node:10938) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'notifyFns' of undefined
    at complete (/usr/share/nodejs/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:414:36)
    at /usr/share/nodejs/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:427:21
(node:10938) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:10938) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
DFU mode set, reconnecting...
Using firmware: ble_app_sensor_humitemp_pca10040e_s112_jo_0618.bin
Unable to find DFU service
```

Any suggestions ?

pacmac, 2020-04-19:
Well I tried this again, and unfortunatley it has bricked the device and I can no longer connect to it using nrfConnect on my phone.

The "Updating application" restarted 5 or 6 times, and then finally hung at 14%, thereafter the device is bricked.

```
connecting...
(node:11351) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'notifyFns' of undefined
    at complete (/usr/share/nodejs/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:414:36)
    at /usr/share/nodejs/node_modules/web-bluetooth-dfu/lib/secure-dfu.js:427:21
(node:11351) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:11351) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
DFU mode set, reconnecting...
Using firmware: ble_app_sensor_humitemp_pca10040e_s112_jo_0618.bin
Updating application [===                 ] 14% 157.4s
```

thegecko, 2020-04-19:
Which version of Node.js?

pacmac, 2020-04-19:
Hi, the node version is v10.16.3.

thegecko, 2020-04-19:
This hasn't been tested on Node 10.x AFAIK. I'd be interested to know if your problem still exists with Node 8.x.

unaisqb, 2020-04-27:
@pacmac 
Please specify  optionalServices:[0xFE59] in request device method. 0xFE59 is default DFU service UUID. https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth/requestDevice

closed 2020-07-22 by thegecko