thegecko/web-bluetooth-dfu #49: nRF51 Support

septs, 2020-09-05:
see https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v4.x.x/doc/html/group__dfu__ble__service__spec.html

thegecko, 2020-09-05:
This [supports nRf51 with softdevice S130](https://github.com/thegecko/web-bluetooth-dfu#device-configuration)

septs, 2020-09-05:
[image]

the target device DFU use 00001530-1212-efde-1523-785feabcd123

thegecko, 2020-09-05:
I assume the default has changed, overwrite it here: https://github.com/thegecko/web-bluetooth-dfu/blob/d050b89497ae297829a772788e0c052781f4b945/src/secure-dfu.ts#L458

septs, 2020-09-05:
OK.

septs, 2020-09-05:
```
ERROR in ./src/components/SensorConsole/dfu/index.ts
Module not found: Error: Can't resolve 'web-bluetooth-dfu' in '/Users/septs/Projects/mops-vida-pm-watchdog/src/components/SensorConsole/dfu'
 @ ./src/components/SensorConsole/dfu/index.ts 2:0-42 6:20-29 7:25-34
 @ ./src/components/SensorConsole/FirmwareVersion.tsx
 @ ./src/components/SensorConsole/index.tsx
 @ ./src/Entry.tsx
 @ ./src/index.ts
```

septs, 2020-09-06:
https://github.com/thegecko/web-bluetooth-dfu/blob/5cdb47563df3d2188d0f1051b9954022f56dcbd6/package.json#L10-L12
this syntax on webpack may not be supported.

Reference: https://webpack.js.org/configuration/resolve/

septs, 2020-09-07:
@thegecko

thegecko, 2020-09-07:
This is an old repo and wasn't designed with webpack in mind. PRs welcome.

septs, 2020-09-07:
@thegecko do i allow many changes? e.q:
upgrade all packages
migrate tslint to eslint
etc.

thegecko, 2020-09-07:
Please keep separate PRs focussed on a single area

septs, 2020-09-07:
Sorry, I decided to write a more modern
https://github.com/NiceLabs/nordic-dfu

closed 2020-09-07 by septs