thegecko/web-bluetooth-dfu #53: DFU is very slow

Bugliaa, 2020-09-14:
The update process actually works but it's very slow, it takes about 100 seconds to transfer a 70 kB image to the nRF52832 device. Is there anything I can do to speed up the process? I've already tried checking the MTU size but it seems ok, moreover using the same MTU to exchange files via OBEX I can obtain a much faster tranfer.

Thank you in advance!

thegecko, 2020-09-14:
Not as far as I know. BLE has very small packets in the range of 10s of bytes. Add in the frame headers and the data transfer is never going to be quick!

closed 2021-02-19 by thegecko

fanoush, 2021-03-29:
In my experience (with nordic nrf connect for android and python code https://github.com/fanoush/dsd6-ota-dfu-python/blob/master/ble_secure_dfu_controller.py ) speeds like 2.5KB/s should be doable without any MTU changes. Disabling packet receipt notifications helps but I see it is not even used/enabled in this code. Also with nrfconnect for android it helps to increase connection priority when uploading to bootloader based on SDK14 and up, otherwise speed is below 1KB/s. not sure if this could be done in web bluetooth? what it does is to pick shorter connection interval
https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/master/documentation/Automated%20tests/README.md#request-connection-priority-change
With SDK12 based bootloader this in not needed, it is high even without it.