thegecko/web-bluetooth-dfu #56: "this" is undefined when setting dfu mode

xsorifc28, 2021-02-16:
Using Chrome:
Google Chrome | 88.0.4324.150 (Official Build) (x86_64)
-- | --
Revision | e6c2a846b660e61f059daee2c46217af42913240-refs/branch-heads/4324@{#2103}
OS | macOS Version 11.1 (Build 20C69)
JavaScript | V8 8.8.278.15
User Agent | Mozilla/5.0 (Macintosh; Intel Mac OS X 11_1_0)

The following line throws a null pointer:
https://github.com/thegecko/web-bluetooth-dfu/blob/master/src/secure-dfu.ts#L497

One way around this is using arrow syntax `() => {...}` to scope "this", but I'm unsure why the error is happening in the first place?

thegecko, 2021-02-16:
Does this happen when using the live example?

xsorifc28, 2021-02-18:
Yes, I reproduced this with the live example.

thegecko, 2021-02-19:
Thanks for noticing this, should now be fixed in https://github.com/thegecko/web-bluetooth-dfu/commit/02d64e17af0b2f34d3aaa8aa0c3decc3b033e6e9

closed 2021-02-19 by thegecko