thegecko/web-bluetooth-dfu #80: "Error: The state of the DFU process does not allow this operation" in (some) Windows 11 Web Bluetooth

gfwilliams, 2024-01-10:
Hi! I'm aware this repo hasn't been updated in a while but I thought this was worth reporting... All I can see is a reference to the error [in an issue related to packet size](https://github.com/thegecko/web-bluetooth-dfu/issues/36).

To reproduce, on Windows 11:

* Go to https://thegecko.github.io/web-bluetooth-dfu/examples/web.html
* Select a DFU zip (using Espruino for now)
* Put Espruino device in DFU mode
* Select the `DfuTarg` device

The updater *appears* to work for a second - the progress bar increases gradually up to 4096 so it's sending multiple packets, but then it fails with `Error: The state of the DFU process does not allow this operation` (NRF_DFU_RES_CODE_OPERATION_NOT_PERMITTED). Unfortunately at that point it's already erased the app, so the user is left with a nonfunctional device.

On PCs where it happens, it happens repeatably, every time.

Refreshing the page and trying again seems to complete on 2nd/3rd attempts, but there's usually a "Data object does not match the firmware..." (NRF_DFU_RES_CODE_INVALID_OBJECT) message, and sometimes the uploaded firmware is corrupt.

This seems to have been an issue for a while but I had another report recently and it seems I now have a PC that exhibits it. So far my testing:

* **FAIL:** Lenovo Ideapad 5 Pro, Windows 11 (Realtek BT)
* **FAIL:** Lenovo Legion GO, Windows 11 (Mediatek BT)
* OK: Dell XPS 13 (older variant) works fine on Windows 10  (Intel BT)
* OK: Lenovo Ideapad 530S, Windows 11 (Intel BT)

Log looks like:

```
connected to gatt server
found DFU service
found 2 characteristic(s)
connected to gatt server
found DFU service
found 2 characteristic(s)
found packet characteristic
found control characteristic
enabled control notifications
transferring init
written 144 bytes
transfer complete
transferring firmware
written 0 bytes
notify: Error: The state of the DFU process does not allow this operation
```

Sometimes it even says `written 256 bytes` rather than `written 0 bytes` - but never 4096, which is what it should be.

As far as I can see, to get `written 0 bytes` the CRC has to have matched, but the actual response from DFU gives the wrong value: https://github.com/thegecko/web-bluetooth-dfu/blob/master/src/secure-dfu.ts#L369-L378

I tried patching it to check if `checkCrc(..) && transferred!=0` but this just results in a "Data object does not match the firmware..." (NRF_DFU_RES_CODE_INVALID_OBJECT) message when it retries calling `transferObject`.

Not sure if you've got any thoughts on this? I'm a bit stumped...

thegecko, 2024-01-10:
Your right, Gordon I haven't touched this in years!

My initial thought is the NRF DFU process has probably changed since this was written (v12), but that doesn't explain why it works in some cases.

What browser are you using, I assume chrome?

gfwilliams, 2024-01-10:
Hi Rob, thanks! However I'm still using the same DFU bootloader that I always have so I don't think that's an issue - and yes, it's Chrome (I also just tested on Edge - exactly the same failure).

I've had the odd complaint for years, but always put it down to it being some kind of glitch. It's nice to be in a position where it's at least repeatable now.

I've just been running some tests, and trying to retry sending packets on failure isn't working for me. However setting `delay=5` when creating SecureDfu (which adds a delay between sending packets) **does** fix it.

I did some *more* fiddling, and found I could change the error handler for `update` to:

```
            .catch(error => {
                if (this.delay === 0) {
                    this.log("DFU update failed, but delay=0. Trying again with delay=10...");
                    this.delay = 10;
                    return this.update(device, init, firmware);
                }
                reject(error);
            });
```

(so to retry from the start, with a delay) and so far this seems reliable. Do you think it's something you'd be interested in having a PR for? It's a bit of a hack, but I guess it's never called unless there's an error.

gfwilliams, 2024-01-11:
Just done!

closed 2024-01-11 by thegecko

thegecko, 2024-01-11:
Do you need a release too?

gfwilliams, 2024-01-11:
No, I'm fine - but I guess it's worth making sure that the GitHub pages version has the change in?

PR #81 (2024-01-11): If DFU update fails, automatically try again but with a delay between writes
If DFU update fails but we're not using any delay between writes, try again but with a delay.

This fixes an issue with *some* Windows 11 PCs where a DFU update would fail on the first 4096 byte block

Fix #80
commit 334e3367 (2024-01-11): If DFU update fails but we're not using any delay between sending packets, try again but with a delay.

This fixes an issue with *some* Windows 11 PCs where a DFU update would fail on the first 4096 byte block

Fix #80